LBP311/Origami-Electronics #36: Touch Sensor "Liveness"

LBP311, 2022-02-09:
Ensure part works as expected.
Include the following for each component:

- [ ] Photo/Schematic of working "liveness"
- [ ] Expected Result
- [ ] Instructions for implementation
- [ ] (If Applicable) Code

LBP311, 2022-02-10:
- Expected Result:
    - Each LED lights up individually when touching the corresponding wire
- Instructions for Implementation
    - VDD = 1.8 to 5.5 VDC
    - GND connected to GND
    - Attach wires for "Touch Sensing" to the connections near GND.
- Recommendations
    - After first connecting the "Touch Sense" wire, wait some time for the capacitor to discharge.
        - When first connected, the corresponding LED will turn on and seem as it is not working when touching and not touching
    - After the capacitor discharges, the LED will turn off
    - Once the LED is off, it will operate as intended
- Extra Experimentation
    - Add resistors in series with the "Touch Sense" wire and the corresponding pin

LBP311, 2022-02-15:
**Revisions/Comments:**
- Instructions for Implementation
    - Immediate Results
        1. Attach wires for "Touch Sensing"
            - These wires will be connected near the GND terminal
        2. Connect GND to GND
        3. Connect VDD to (1.8 to 5.5 VDC)
    - Delayed Results (~30 seconds)
        1. Connect GND to GND
        2. Connect VDD to (1.8 to 5.5 VDC)
        3. Connect "Touch Sensor" wires
            - These wires will be connected near the GND terminal
            - Delay begins after connecting the first "Touch Sensor" wire
- Delayed Results Notes:
    - While the LED is ON (during delay), NONE of the Touch Sensors will respond as expected
    - Once the corresponding LED is OFF, "Touch Sensor" wires will respond as expected
    - If additional wires are connected within the delay:
        - The additional "Touch Sensor" wires will function as expected after delay ends
    - If additional wires are connected after the delay ends:
        - The delay will commence again 

- Extra Experimentation (No Longer Required)

LBP311, 2022-02-15:
**Note:** If a charged wire touches another wire, you may experience undesired results. Ensure that each wire is separate from other wires/components that could provide a charge to a "Touch Sensor" wire.

LBP311, 2022-02-15:
**Photo/Schematic of working "Liveliness"**

Adding Additional Wires during Delay Period:
https://user-images.githubusercontent.com/98776496/153996013-3192c87e-0275-44dd-bbb8-f54dc8cda273.mp4

Introducing Delay:
https://user-images.githubusercontent.com/98776496/153996525-76d021bc-b592-4747-82a7-a7c2b9554353.mp4

Immediate Results:
https://user-images.githubusercontent.com/98776496/153996850-ad3af7db-e393-494e-a518-8904d4aa892f.mp4

Expected Results:
https://user-images.githubusercontent.com/98776496/153997035-51142757-c80c-4941-8a03-9642b45a3217.mp4

Continuous Delay:
https://user-images.githubusercontent.com/98776496/153997989-f8ae152d-5ede-4982-8093-7272cb32db0d.mp4

LBP311, 2022-02-15:
**No Code required for Liveliness Testing**

mrm815, 2022-02-23:
Looks good!

LBP311, 2022-02-24:
Issue reopened ONLY to clarify the requirement of Task 1 to ensure proper testing of "liveness".
See Issue #21 comments for further detail.

Issue will be closed again.

closed 2022-02-24 by LBP311